GenPi64/genpi64-overlay #18: Missing packages

ormorph, 2021-01-25:
Missing packages:
```
dev-embedded/rpi4-eeprom-updater
sys-apps/rpi3-expand-swap
sys-firmware/brcm43430-firmware
sys-boot/rpi3-64bit-firmware
dev-embedded/rpi4-eeprom-images
```
Installed only through binhost.
There is also a dependency problem, `media-libs/raspberrypi-userland` is not put, the problem is in the version name. The version is also installed only from binhost.
This is not very good.

samip5, 2021-01-25:
Please be more specifc, which image is this?

ormorph, 2021-01-26:
The image is quite old, just started updating and noticed. This is a dependency in `dev-embedded/rpi-64bit-meta`. If these packages are no longer needed, these dependencies will need to be fixed.
I use it to avoid cluttering @world. This ebuild needs to be fixed, there are many outdated dependencies.

samip5, 2021-01-26:
> The image is quite old, just started updating and noticed. This is a dependency in `dev-embedded/rpi-64bit-meta`. If these packages are no longer needed, these dependencies will need to be fixed.
> I [...]

Okey, so it seems that you're most likely still using sakaki's binhost? Check /etc/portage/make.conf for the BINHOST. The up-to-date one is at `https://packages.genpi64.com`.

Please unmerge `dev-embedded/rpi-64bit-meta` as it's obsolate and no-longer used.

ormorph, 2021-01-26:
I switched to the new binhost long ago.
>Please unmerge dev-embedded/rpi-64bit-meta as it's obsolate and no-longer used.

This brings up a new issue, which packages should now be added to @world. This will most likely break the system.

samip5, 2021-01-26:
> I switched to the new binhost long ago.
> 
> > Please unmerge dev-embedded/rpi-64bit-meta as it's obsolate and no-longer used.
> 
> This brings up a new issue, which packages should now be added to @world. This will most likely break the system.

I included the needed things to add to world sets. The pi4desktop can most likely be omitted, if you're not running the Sakaki full image.

perkinslr, 2021-01-26:
> I switched to the new binhost long ago.
> 
> > Please unmerge dev-embedded/rpi-64bit-meta as it's obsolate and no-longer used.
> 
> This brings up a new issue, which packages should now be added to @world. This will most likely break the system.

It's a fair point that the current state is likely a bit on the fragile side, as people will have to add or remove packages from `/etc/portage/sets/pi4`.  The meta-package is fragile in other ways.  

Profile targets can supply a *single* set file, so we could probably add the sets to profile targets.  Then which sets get included in `@profile` depend on the profile selected.  

The other option is to create a package the manages the sets files explicitly.

None of these options are without issue, but I do want to have *something* figured out for the next release.

ormorph, 2021-01-26:
I mostly use the legacy of @Sakaki's image.  I am building python with the tk flag enabled. I switched to this overlay a long time ago. I collect the kernel myself. It depended on the flags of this package.

perkinslr, 2021-01-26:
So if you look at NeddySeagoon's genpi64-overlay, he's been spearheading the effort to upstream packages.  The packages you list in your initial question have all been successfully integrated into `::gentoo`, with minor name changes (except `expand-swap`, as that's going to need some special care).  The binpkgs are remanents of the first couple alpha releases.  They'll get purged next time the binhost gets a full refresh (likely when the image builds without intervention).

ormorph, 2021-01-26:
I think it will be easier for me to write my own meta package. Actually, I do it, I have a modified metapackage in the local overlay.

perkinslr, 2021-01-26:
If that makes it easier to see what needs updating in your own workflow, that should work for you.  I would strongly suggest using the `::gentoo` packages where possible.  At best you'd otherwise have to maintain them in parallel yourself.  At worst you'll end up running bit-rotting packages.

ormorph, 2021-01-26:
>I would strongly suggest using the ::gentoo
Actually, I was going to do this.

ormorph, 2021-01-26:
Slow packet transfer to Gentoo upstream, and still there was a mismatch. The `media-libs/raspberrypi-userland package` from the Gentoo upstream is not used. [link](https://packages.genpi64.com/media-libs/raspberrypi-userland/).
The version from binhost is still used. It was necessary to mask version 1.20200520. This package is still in dependencies:
>\# wget https://raw.githubusercontent.com/GenPi64/Build.Dist/alpha6/config/sets/pi4

samip5, 2021-01-26:
The binhost will most likely get fresh packages at some time today.

ormorph, 2021-01-26:
By the way, after switching to `raspberrypi-userland` from uppstream, do you have problems building `media-libs/libsdl2`?
I had to add the line to the libsdl2 ebuild:
```
append-ldflags -lvchostif
```
Otherwise, it was not going to and the error popped up: `undefined reference to ...`

samip5, 2021-02-16:
> By the way, after switching to `raspberrypi-userland` from uppstream, do you have problems building `media-libs/libsdl2`?
> I had to add the line to the libsdl2 ebuild:
> 
> ```
> append-ldflags -lvchostif
> ```
> 
> Otherwise, it was not going to and the error popped up: `undefined reference to ...`

I have no idea as I don't run a desktop on my RPi's. Most of them are headless.

ormorph, 2021-02-18:
>I have no idea as I don't run a desktop on my RPi's. Most of them are headless.

I had problems building `media-libs/libsdl2-2.0.14-r2`. Therefore, I had to add this flag for linking. The error occurs during build.

samip5, 2021-03-04:
> > I have no idea as I don't run a desktop on my RPi's. Most of them are headless.
> 
> I had problems building `media-libs/libsdl2-2.0.14-r2`. Therefore, I had to add this flag for linking. The error occurs during build.

We have added a patched libsdl2 to the overlay in #42.

ormorph, 2021-03-05:
Later I will try to collect this ebuild, I will write about the results.

ormorph, 2021-03-05:
>We have added a patched libsdl2 to the overlay in 33eb8bf. 

There are build errors here. Added a [fix](https://github.com/GenPi64/genpi64-overlay/pull/43).

closed 2021-03-05 by ormorph