GenPi64/genpi64-overlay #21: broken distfiles links

gissimo, 2021-01-28:
Hi,
Unfortunately latest [commit](https://github.com/GenPi64/genpi64-overlay/commit/4b305a36fd357799fa8f6f45230b3123a0fc0d7f) breaks download of some distfiles, such as [here](https://github.com/GenPi64/genpi64-overlay/blob/master/sys-kernel/bcm2711-kernel-bis-bin/bcm2711-kernel-bis-bin-5.4.69.20201006.ebuild).
Seems to be due to fact that not all releases are ready under GenPi64/genpi64-overlay whereas they were under sakaki-/genpi64-overlay, e.g. [this](https://github.com/GenPi64/bcm2711-kernel-bis/releases/download/5.4.69.20201006/bcm2711-kernel-bis-5.4.69.20201006.tar.xz) gives 404.
Thanks,
G

perkinslr, 2021-01-29:
Hm, good catch, hadn't thought about forks not getting the old release snapshots...  Probably need to go through the ebuilds and determine which are affected.  Of those affected, we should do one of the following.

1. Create a release snapshot at the specified URL.
2. Point the URL back to Sakaki's version
3. Remove the ebuild.

For a start, all be the latest kernel versions should go away.  The latest one should point at Sakaki's version.  Once we're reasonably confident in the raspberrypi-kernel package, all the other kernel packages should go away.

ormorph, 2021-01-29:
Same with `sys-apps/pyconfig_gen`, wrong url. Since this package was in `distfiles`, I did not notice that the link was not working. Another option to download manually from @sakaki- in `distfiles`.

samip5, 2021-04-01:
Please make a new bug report if the distfiles are still not correct.

closed 2021-04-01 by samip5