GenPi64/genpi64-overlay #33: Issues with PYTHON_COMPAT

jonesmz, 2021-02-13:
uilding database (/var/cache/eix/portage.eix)...
[0] "gentoo" /var/db/repos/gentoo/ (cache: metadata-md5-or-flat)
     Reading category 167|167 (100) Finished
[1] "genpi-tools" /var/db/repos/sakaki-tools (cache: parse|ebuild*#metadata-md5#metadata-flat#assign)
     Reading category 167|167 (100) Finished
[2] "genpi64" /var/db/repos/genpi64 (cache: parse|ebuild*#metadata-md5#metadata-flat#assign)
     Reading category  35|167 ( 20): dev-embedded... * ERROR: dev-embedded/pitop-speaker-1.1.0.1-r2::genpi64 failed (depend phase):
 *   No supported implementation in PYTHON_COMPAT.
 *
 * Call stack:
 *                         ebuild.sh, line 609:  Called source '/var/db/repos/genpi64/dev-embedded/pitop-speaker/pitop-speaker-1.1.0.1-r2.ebuild'
 *   pitop-speaker-1.1.0.1-r2.ebuild, line   8:  Called inherit 'distutils-r1' 'versionator'
 *                         ebuild.sh, line 314:  Called __qa_source '/var/db/repos/gentoo/eclass/distutils-r1.eclass'
 *                         ebuild.sh, line 112:  Called source '/var/db/repos/gentoo/eclass/distutils-r1.eclass'
 *               distutils-r1.eclass, line 105:  Called inherit 'python-r1'
 *                         ebuild.sh, line 314:  Called __qa_source '/var/db/repos/gentoo/eclass/python-r1.eclass'
 *                         ebuild.sh, line 112:  Called source '/var/db/repos/gentoo/eclass/python-r1.eclass'
 *                  python-r1.eclass, line 260:  Called _python_set_globals
 *                  python-r1.eclass, line 199:  Called _python_set_impls
 *            python-utils-r1.eclass, line 156:  Called die
 * The specific snippet of code:
 *   		                die "No supported implementation in PYTHON_COMPAT."
 *
 * If you need support, post the output of `emerge --info '=dev-embedded/pitop-speaker-1.1.0.1-r2::genpi64'`,
 * the complete build log and the output of `emerge -pqv '=dev-embedded/pitop-speaker-1.1.0.1-r2::genpi64'`.
 * Working directory: '/usr/lib/python3.8/site-packages'
 * S: '/pitop-speaker-1.1.0.1'

ebuild failed with status 1
     Reading category  35|167 ( 20): dev-embedded...
cannot properly execute /var/db/repos/genpi64/dev-embedded/pitop-speaker/pitop-speaker-1.1.0.1-r2.ebuild
     Reading category 111|167 ( 66): net-libs... * ERROR: net-libs/nodejs-9.11.2-r2::genpi64 failed (depend phase):
 *   No supported implementation in PYTHON_COMPAT.
 *
 * Call stack:
 *                 ebuild.sh, line 609:  Called source '/var/db/repos/genpi64/net-libs/nodejs/nodejs-9.11.2-r2.ebuild'
 *   nodejs-9.11.2-r2.ebuild, line   9:  Called inherit 'bash-completion-r1' 'eutils' 'flag-o-matic' 'pax-utils' 'python-single-r1' 'toolchain-funcs'
 *                 ebuild.sh, line 314:  Called __qa_source '/var/db/repos/gentoo/eclass/python-single-r1.eclass'
 *                 ebuild.sh, line 112:  Called source '/var/db/repos/gentoo/eclass/python-single-r1.eclass'
 *   python-single-r1.eclass, line 269:  Called _python_single_set_globals
 *   python-single-r1.eclass, line 209:  Called _python_set_impls
 *    python-utils-r1.eclass, line 156:  Called die
 * The specific snippet of code:
 *   		                die "No supported implementation in PYTHON_COMPAT."
 *
 * If you need support, post the output of `emerge --info '=net-libs/nodejs-9.11.2-r2::genpi64'`,
 * the complete build log and the output of `emerge -pqv '=net-libs/nodejs-9.11.2-r2::genpi64'`.
 * Working directory: '/usr/lib/python3.8/site-packages'
 * S: '/nodejs-9.11.2'

ebuild failed with status 1
     Reading category 111|167 ( 66): net-libs...
cannot properly execute /var/db/repos/genpi64/net-libs/nodejs/nodejs-9.11.2-r2.ebuild
     Reading category 111|167 ( 66): net-libs... * ERROR: net-libs/nodejs-9.11.2-r4::genpi64 failed (depend phase):
 *   No supported implementation in PYTHON_COMPAT.
 *
 * Call stack:
 *                 ebuild.sh, line 609:  Called source '/var/db/repos/genpi64/net-libs/nodejs/nodejs-9.11.2-r4.ebuild'
 *   nodejs-9.11.2-r4.ebuild, line   9:  Called inherit 'bash-completion-r1' 'eutils' 'flag-o-matic' 'pax-utils' 'python-single-r1' 'toolchain-funcs'
 *                 ebuild.sh, line 314:  Called __qa_source '/var/db/repos/gentoo/eclass/python-single-r1.eclass'
 *                 ebuild.sh, line 112:  Called source '/var/db/repos/gentoo/eclass/python-single-r1.eclass'
 *   python-single-r1.eclass, line 269:  Called _python_single_set_globals
 *   python-single-r1.eclass, line 209:  Called _python_set_impls
 *    python-utils-r1.eclass, line 156:  Called die
 * The specific snippet of code:
 *   		                die "No supported implementation in PYTHON_COMPAT."
 *
 * If you need support, post the output of `emerge --info '=net-libs/nodejs-9.11.2-r4::genpi64'`,
 * the complete build log and the output of `emerge -pqv '=net-libs/nodejs-9.11.2-r4::genpi64'`.
 * Working directory: '/usr/lib/python3.8/site-packages'
 * S: '/nodejs-9.11.2'

ebuild failed with status 1
     Reading category 111|167 ( 66): net-libs...
cannot properly execute /var/db/repos/genpi64/net-libs/nodejs/nodejs-9.11.2-r4.ebuild
     Reading category 111|167 ( 66): net-libs... * ERROR: net-libs/nodejs-9.11.2-r6::genpi64 failed (depend phase):
 *   No supported implementation in PYTHON_COMPAT.
 *
 * Call stack:
 *                 ebuild.sh, line 609:  Called source '/var/db/repos/genpi64/net-libs/nodejs/nodejs-9.11.2-r6.ebuild'
 *   nodejs-9.11.2-r6.ebuild, line   9:  Called inherit 'bash-completion-r1' 'eutils' 'flag-o-matic' 'pax-utils' 'python-single-r1' 'toolchain-funcs'
 *                 ebuild.sh, line 314:  Called __qa_source '/var/db/repos/gentoo/eclass/python-single-r1.eclass'
 *                 ebuild.sh, line 112:  Called source '/var/db/repos/gentoo/eclass/python-single-r1.eclass'
 *   python-single-r1.eclass, line 269:  Called _python_single_set_globals
 *   python-single-r1.eclass, line 209:  Called _python_set_impls
 *    python-utils-r1.eclass, line 156:  Called die
 * The specific snippet of code:
 *   		                die "No supported implementation in PYTHON_COMPAT."
 *
 * If you need support, post the output of `emerge --info '=net-libs/nodejs-9.11.2-r6::genpi64'`,
 * the complete build log and the output of `emerge -pqv '=net-libs/nodejs-9.11.2-r6::genpi64'`.
 * Working directory: '/usr/lib/python3.8/site-packages'
 * S: '/nodejs-9.11.2'

perkinslr, 2021-02-14:
I believe nodejs from the overlay is outdated (and I thought there was a PR to remove it).  I *know* the pitop-speaker package should get mothballed until someone with a pitop shows up to test it.

samip5, 2021-02-16:
I'm going to include the removal of nodejs from overlay with my obsolate packages PR.

samip5, 2021-02-16:
Oh, actually. It seems that nodejs already is removed, so I will close this.

closed 2021-02-16 by samip5