GenPi64/genpi64-overlay #34: Wrong sys-kernel/raspberrypi-kernel version installed

enx1998, 2021-02-16:
**Describe the bug**
I have installed `=sys-kernel/raspberrypi-kernel-5.10.2` using binaries but `uname -r` returns me `5.10.0-rc5-v8` and modules are installed in `/lib/modules/5.10.0-rc5-v8` folder.

**To Reproduce**
Steps to reproduce the behavior:
1. Run `sudo emerge -va sys-kernel/raspberrypi-kernel` with the following use flag `bcm2711 bcmrpi3 -bcm2709 -bcmrpi -debug -initramfs -savedconfig -test`

**Expected behavior**
Kernel 5.10.2 installed, `uname -r` showing `5.10.2` and modules installed inside `/lib/modules/5.10.2` folder.

**Image details (please complete the following information):**
- Version used: migrated from sakaki installation

perkinslr, 2021-02-16:
It's a known problem, with no clear solution...  The problem is we're using the Pi Foundation kernel source, which is version `5.10.y`, meaning 5.10.0 with whatever random patches they pull in from upstream 5.10.  Gentoo picks the specific version number by pulling in `X.Y.0` and then a patchset to get to `X.Y.Z`.  If we try the same, starting with the Pi Fondation sources, we get patches which *partially* apply.  I'd just set the version information on the ebuild to 5.10.y, but that isn't a valid gentoo version number.  

Figuring out *something* is on the list of things to do with the current release.

perkinslr, 2021-02-16:
Looks like if we use the tagged versions off 
https://github.com/raspberrypi/linux/tags
we get sane version numbers.  We'll have to manually map the tag to the version number, as in the source URI within the ebuild won't match the ebuild name in any way (tag 1.20210201-1 is 5.10.11), but at least it gets us, possibly, sane ebuilds...

enx1998, 2021-02-20:
Hi perkinslr,
thanks for your reply, I will wait new ebuild.
Thanks for your work

perkinslr, 2021-03-05:
Should be good to go, mostly.  `5.10.11` actually installs `5.10.11`.  Just note that it isn't slotted properly, so it'll try to install alongside `5.10.0` and collide.

closed 2021-04-28 by perkinslr