GenPi64/genpi64-overlay #59: [question] pigpio dependencies

raballew, 2021-08-30:
**Question**

I am running a custom build of Kinvolks Flatcar Container Linux with some adjustments to https://github.com/kinvolk/coreos-overlay and containing the following:

- Enabled CONFIG_GPIO_SYSFS kernel option, so that /sys/class/gpio/... is present, and /dev/gpiomem is present
- Add pigpio  (with some fixes to make it work on Flatcar)

When starting pigpiod, I am getting errors similar to https://github.com/joan2937/pigpio/issues/461 saying that I can not open a messagebox (mbox). The command `sudo modprobe vcio` fails, as the module is not present on the machine. I assume, that I have to add an ebuild or two, but can not figure out which ones to add. 

At this stage I am just guessing and could use some guidance. This repository somehow got pigpio  working on gentoo and I was wondering, what it took to do so? My next guess would be to add raspberrypi-userland but this seems not be part of this repository anymore, so any ideas or any help is welcome.

jonesmz, 2021-08-30:
This project inherits from a previous project, where the sole-developer of the previous project needed to abandon their work due to career related decisions.

We don't have any meaningful contact with the original project developer, but you might be able to reach out to them and ask.

That being said, I'm not seeing pigpio in our overlay.

Could you link me to the ebuild so I can try to provide some feedback?

jonesmz, 2021-08-30:
Oh. Nevermind. I found it: https://github.com/GenPi64/genpi64-overlay/blob/master/dev-libs/pigpio/pigpio-71-r1.ebuild

jonesmz, 2021-08-30:
I don't see any dependencies listed for pigpio. All it has are two patches that are applied to the pigpio code before installation.

https://github.com/GenPi64/genpi64-overlay/blob/master/dev-libs/pigpio/files/pigpio-71-makefile.patch
https://github.com/GenPi64/genpi64-overlay/blob/master/dev-libs/pigpio/files/pigpio-71-netcomms-arm64-pt2.patch

That's basically all I can suggest for you to try.

Good luck.

raballew, 2021-08-31:
@jonesmz Thanks for the prompt reply. I did a litte more research and found that pigpio directly uses feature provided by Raspberrys Broadcoms VideoCore. Maybe you know which ebuild provides this functionality?

jonesmz, 2021-08-31:
That'd be the device tree overlay files, wouldn't it?

Probably comes from the RPI foundations kernel repository.

raballew, 2021-08-31:
@jonesmz Probably! Taking this, the information is basically provided at https://github.com/raspberrypi/firmware/tree/master/boot/overlays

I will try to go with https://packages.gentoo.org/packages/sys-boot/raspberrypi-firmware first or I am going to replace the Flatcar CoreOS kernel with the original one of Raspberry. 

Anyway thanks for the help!

jonesmz, 2021-08-31:
You're welcome.

You're also welcome to join us on our discord channel for other discussions in the future: https://discord.gg/YvAQK4mT

closed 2021-08-31 by jonesmz

raballew, 2021-09-15:
@jonesmz The invitation links seems to be expired. Could you share a fresh one?

jonesmz, 2021-09-15:
https://discord.gg/Eu5D2e5KJQ

raballew, 2021-09-15:
@jonesmz It seems that this one is invalid too :/ Try adding raballew#6171 directly.

jonesmz, 2021-09-15:
Try one more time. Generated it just now https://discord.gg/423J2FBs