GenPi64/genpi64-overlay #74: sys-firmware/bcm4340a1-firmware is broken

gagara, 2022-01-11:
**Describe the bug**
emerge of sys-firmware/bcm4340a1-firmware is failed

**To Reproduce**
Steps to reproduce the behavior:
1. emerge sys-firmware/bcm4340a1-firmware

**Expected behavior**
emerge should be OK

**Additional context**
emerge failed with
```
FileNotFoundError: [Errno 2] No such file or directory: b'BCM43430A1.hcd'
```

jonesmz, 2022-01-18:
The ebuild for this is pretty simple.

It just downloads

   wget https://raw.githubusercontent.com/RPi-Distro/bluez-firmware/8445a53ce2c51a77472b908a0c8f6f8e1fa5c37a/broadcom/BCM43430A1.hcd

and then installs that file.

Are you able to download that file on yuor system?

gagara, 2022-01-18:
It's weird, but I have
```
>>> Preparing source in /var/tmp/portage/sys-firmware/bcm4340a1-firmware-1.2/work/bcm4340a1-firmware-1.2 ...
--2022-01-18 12:35:17--  https://raw.githubusercontent.com/RPi-Distro/bluez-firmware/8445a53ce2c51a77472b908a0c8f6f8e1fa5c37a/broadcom/BCM43430A1.hcd
Resolving raw.githubusercontent.com... failed: Temporary failure in name resolution.
wget: unable to resolve host address ‘raw.githubusercontent.com’
>>> Source prepared.
```
however I can download it manually.
@jonesmz are you sure that `wget`-ing resource is the right way? Shouldn't we place it in `SRC_URI` list instead?

jonesmz, 2022-01-18:
> @jonesmz are you sure that wget-ing resource is the right way? Shouldn't we place it in SRC_URI list instead?

It's *absolutely* not the right way.

It's just how it was done from before genpi64 tookover the work of sakaki.

I'd be happy to work with you on a new version of the ebuild, if you're interested in helping out with that.

gagara, 2022-01-19:
It appears that it is not possible to use it in `SRC_URI` while we use `git-r3` eclass (at least I didn't find proper way).
So, I propose to place that `hcd` file in `${FILESDIR}` and copy it during prepare phase (in similar way how we do with patches)
See PR for details. Tested on my side. Works fine.

samip5, 2022-01-19:
> It's _absolutely_ not the right way.

It's actually me who did it like that... Because I had no idea how to do it otherwise.

jonesmz, 2022-01-23:
Fixed via https://github.com/GenPi64/genpi64-overlay/pull/82

closed 2022-01-23 by jonesmz