GenPi64/genpi64-overlay #84: Sdl2 package from gentoo wont build without patch

jonesmz, 2022-01-21:
**Describe the bug**
<-- A clear and concise description of what the bug is. -->

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
<-- A clear and concise description of what you expected to happen. -->

**Screenshots**
<-- If applicable, add screenshots to help explain your problem. -->

**Image details (please complete the following information):**
- Version used: <-- e.g. alpha6 -->

**Additional context**
<-- Add any other context about the problem here. -->

samip5, 2022-01-27:
Any ideas of what kind of patch is required?

jonesmz, 2022-01-27:
Patch is already in overlay, but the new sdl2 version has a big difference in ebuild code

k9spud, 2022-05-08:
I just compiled `media-libs/libsdl2-2.0.20-r1` from the Gentoo repository and it finally worked on my Raspberry Pi 4 straight up. So, this issue may be old news now.

jonesmz, 2022-05-08:
Thank you for the information.

Can you make a pull request that removes the defunct package?

closed 2022-05-31 by jonesmz