GenPi64/genpi64-overlay #89: sys-kernel/raspberrypi-kernel should use raspberrypi_os's kernel config instead of fedora's

jonesmz, 2022-02-05:
I'm fairly certain that this is the repository that contains the image builder for raspberry pi os, including the 64bit image that was recently released with support for the rpi 0 w 2.

https://github.com/RPi-Distro/pi-gen

We should use their kernel configuration, instead of fedora's, as the baseline for our kernel builds.

samip5, 2022-02-07:
We want Gentoo patches which is why we are following Fedora as those come directly from the gentoo kernel ebuild.

jonesmz, 2022-02-07:
Something about the way we are building our kernel prevents raspberry pi 0w2 from working.

samip5, 2022-02-07:
Unfortunately, that's one if the hard to come by Pis atm so I cannot even test.  :)

jonesmz, 2022-02-07:
Simply looking at the contents of /boot for the official raspberry pi 64 bit image shows that we are missing all sorts of stuff, including the pi0w2 dtbo files

samip5, 2022-02-07:
The missing dtbo files migh be an upstream gentoo problem or where do they come from?

jonesmz, 2022-02-07:
They are built by the kernel build. 

See this conversation https://github.com/RPi-Distro/pi-gen/issues/586