GenPi64/genpi64-overlay #95: dev-embedded/rpi-imager: Revbump to 1.7.1

jonesmz, 2022-02-06:
Current ebuild is v1.6.
Latest is https://github.com/raspberrypi/rpi-imager/releases/tag/v1.7.1

ShorTie8, 2022-07-04:
The latest is now 1.7.2
Here is a working updated ebuild for it
# Copyright 2019 Gentoo Authors
# Distributed under the terms of the GNU General Public License v2

EAPI=7

inherit cmake flag-o-matic

CMAKE_MAKEFILE_GENERATOR="emake"
CMAKE_IN_SOURCE_BUILD=1

DESCRIPTION="Raspberry Pi Imager (WIP ebuild)"
HOMEPAGE="https://github.com/raspberrypi/rpi-imager"
SRC_URI="https://github.com/raspberrypi/rpi-imager/archive/v${PV}.tar.gz"

S="${WORKDIR}/rpi-imager-${PV}/src"

LICENSE="Apache"
SLOT="0"
KEYWORDS="~arm64 ~amd64"
IUSE="+qt5"

DEPEND=""
RDEPEND="${DEPEND}"
BDEPEND="dev-util/cmake 
		 net-misc/curl 
		 app-arch/libarchive 
		 dev-libs/openssl 
		 dev-qt/qtcore 
		 dev-qt/qtsvg 
		 dev-qt/linguist 
		 dev-qt/linguist-tools 
		 dev-qt/qtconcurrent 
		 dev-qt/qtquickcontrols2"

src_prepare() {
    cmake_src_prepare
}

src_configure() {
    local mycmakeargs=(
    )
    cmake_src_configure
}
``

jonesmz, 2022-07-04:
Could I trouble you to make this into a pull request that includes updating the metadata for the new ebuild?

ShorTie8, 2022-07-04:
Sure, but might take awhile to figure it all out
as a quicky you could do 
`cd /var/db/repos/genpi64/dev-embedded/rpi-imager
wget https://dpaste.com/DXU2Y7GR9.txt -O rpi-imager-1.7.2.ebuild
dos2unix rpi-imager-1.7.2.ebuild
ebuild rpi-imager-1.7.2.ebuild manifest
emerge rpi-imager`

jonesmz, 2022-07-11:
I'm happy to wait until you have time.

If I need to, I can integrate your new ebuild, but will probably spend my project time on improving the base image first.

ShorTie8, 2022-07-11:
like too python3_10 maybe ??
As I can not set the profile to genpi64/desktop and update world

jonesmz, 2022-07-11:
That was merged today.

https://github.com/GenPi64/genpi64-overlay/commit/ad0a729b4c456105b92c04da6c8e82bc05707d94

closed 2022-09-02 by jonesmz