GenPi64/genpi64-overlay #144: dev-python/rpi-gpio

ShorTie8, 2022-08-23:
When I try to emerge dev-python/rpi-gpio, I get
`202 /var/db/repos/genpi64 # emerge -pv dev-python/rpi-gpio

These are the packages that would be merged, in order:

Calculating dependencies \

!!! Problem resolving dependencies for dev-python/rpi-gpio                                                                                                                                                               ... done!

!!! The ebuild selected to satisfy "dev-python/rpi-gpio" has unmet requirements.
- dev-python/rpi-gpio-0.7.0::genpi64 USE="" PYTHON_TARGETS="-python3_8 -python3_9"

  The following REQUIRED_USE flag constraints are unsatisfied:
    any-of ( python_targets_python3_8 python_targets_python3_9 )

`

Thankz
Jeff

samip5, 2022-08-23:
Not really an bug when not newer than 3.9 is supported with it.

closed 2022-08-23 by samip5

ShorTie8, 2022-08-26:
If your interested ??
So I took the rpi-64bit-meta ebuild
And instead of deleting it, I riped lxde out in to it's own thing.
Also put all the pi stuff together. Thus needing your genpi64-overlay.
Then added a kde-plazma option.

So it should work on like every thing .. :/~
https://gist.github.com/ShorTie8/659dba5034c5af0f07f01d24570376a2
Needs some cleanup, lol.

Have a Great Day
Jeff

PR #145 (2022-08-23): dev-python/rpi-gpio: Update to support Python 3.10.
Fixes #144