GenPi64/genpi64-overlay #157: Multiple issues with sys-kernel/raspberrypi-kernel

mmajchr, 2022-10-16:
1) For some time (at least on version 5.10.x) default root filesystem for your kernels was btrfs. I flashed my boards around that time. Unfortunately for new releases (confirmed with raspberrypi-kernel-5.15.61) default filesystem is ext and btrfs as module which causes images built with sys-kernel/raspberrypi-kernel not boot at all on my boards. After I ctrl+z emerge and manually changed the config to include btrfs support compiled in instead of module those images boot fine on my boards
2) Second issue is that the port has a savedconfig use flag but unfortunately it's completely ignored :( The config is copied /etc/portage/savedconfig but it's overwritten in work directory during compilation with fresh one from fedora's repo
3) Couldn't you add something like "menuconfig" use flag so people could adjust kernel components during emerge process? I understand that something similar could be achieved with "savedconfig" use flag (once it's fixed of course :)) however adjusting config is often more complicated than changing m->y ;) Having full menuconfig with all it docs, dependencies etc. at hand could be useful :) Of course I understand that's possible to have custom build directory when you can do menuconfig and then just copy it to /etc/portage/savedconfig/sys-kernel/ . Nevertheless having it as part of ebuild makes it more user friendly IMHO.
Hope this helps :)

jonesmz, 2022-10-16:
The kernel is supposed to be building with the contents of /etc/kernel/config.d/*** applied to it.

Our build system has:

```
btrfs.config  docker.config  efi.config  freq_governer.config  gpt.config  hugepages.config  initramfs.config  proc-config.config  zstd_for_zram.config
```

Which comes out to

```
CONFIG_BTRFS_FS=y
CONFIG_BTRFS_FS_POSIX_ACL=y
CONFIG_CGROUP_HUGETLB=y
CONFIG_CGROUP_MEMORY=y
CONFIG_EFI=y
CONFIG_EFI_STUB=y
CONFIG_CPU_FREQ_GOV_PERFORMANCE=y
CONFIG_CPU_FREQ_GOV_POWERSAVE=y
CONFIG_CPU_FREQ_GOV_USERSPACE=y
CONFIG_CPU_FREQ_GOV_ONDEMAND=y
CONFIG_CPU_FREQ_GOV_CONSERVATIVE=y
CONFIG_CPU_FREQ_GOV_SCHEDUTIL=y
CONFIG_CPU_FREQ_DEFAULT_GOV_SCHEDUTIL=y
CONFIG_EFI_PARTITION=y
TRANSPARENT_HUGEPAGE=y
TRANSPARENT_HUGEPAGE_ALWAYS=y
CONFIG_HUGETLBFS=y
HUGETLB_PAGE_FREE_VMEMMAP=y
CONFIG_ARCH_ENABLE_HUGEPAGE_MIGRATION=y
BLK_DEV_INITRD=y
RD_ZSTD=y
CONFIG_IKCONFIG=y
CONFIG_IKCONFIG_PROC=y
CONFIG_ZSWAP_COMPRESSOR_DEFAULT_ZSTD=y
CONFIG_ZSWAP_COMPRESSOR_DEFAULT="zstd"
```

You're saying this isn't working for you?

Really our kernel configuration is just a pretty thin wrapper around the kernel-build eclass, which you can see here: https://github.com/GenPi64/genpi64-overlay/blob/master/eclass/pikernel-build.eclass

The code to merge the .config files is near the bottom, and really the whole thing is a messy hack.

I'd be happy to see some help with this, if you have the time.

jonesmz, 2022-10-16:
For what it's worth, much newer builds are available from our continuous integration system now: 

https://ci.genpi64.com/
https://s3.genpi64.com/images/nullGenPi64OpenRC-13-10-22.img.zst
https://s3.genpi64.com/images/nullGenPi64Systemd-13-10-22.img.zst

mmajchr, 2022-11-30:
Sorry but I was little busy. Yes I do have /etc/kernel/config.d/ with btrfs.config but it's not the one that's being compiled. Maybe emerge switches to initramfs.config? Anyway savedconfig flag is misleading as you copy your config to /etc/portage/savedconfig/sys-kernel/ but do not use it during compilation :) I think if there actually IS a savedconfig it's the one you should prefer. If you want I can try helping with ebuild but it's up to you to decide on the procedure :) Also looks you "merging" .config from stuff from   /etc/kernel/config.d/ ? As configs I have there look "bare" :)
```
 cat /etc/kernel/config.d/initramfs.config 
BLK_DEV_INITRD=y
RD_ZSTD=y
 ~ # cat /etc/kernel/config.d/
btrfs.config          freq_governer.config  gpt.config            initramfs.config      
 ~ # cat /etc/kernel/config.d/freq_governer.config 
CONFIG_CPU_FREQ_GOV_PERFORMANCE=y
CONFIG_CPU_FREQ_GOV_POWERSAVE=y
CONFIG_CPU_FREQ_GOV_USERSPACE=y
CONFIG_CPU_FREQ_GOV_ONDEMAND=y
CONFIG_CPU_FREQ_GOV_CONSERVATIVE=y
CONFIG_CPU_FREQ_GOV_SCHEDUTIL=y
CONFIG_CPU_FREQ_DEFAULT_GOV_SCHEDUTIL=y
 ~ # cat /etc/kernel/config.d/gpt.config 
CONFIG_EFI_PARTITION=y
```

jonesmz, 2022-12-03:
> f you want I can try helping with ebuild but it's up to you to decide on the procedure

Our ebuild is almost completely identical to the Gentoo kernel ebuild, but there certainly might be mistakes in it.

I'd be very pleased to see improvements.

>  Also looks you "merging" .config from stuff from /etc/kernel/config.d/ ? As configs I have there look "bare" :)

It may be that the kernel configuration files aren't being copied over. Take a look at https://github.com/GenPi64/Build.Dist/tree/master/config/pyconfig for the config files, which are used to do initial file copies into the image

S4enL0ng, 2022-12-29:
Because this is a "Multiple issues with sys-kernel/raspberrypi-kernel" Thread.

How about the EAPI?
Upstream gentoo tree has updated the kernel-build eclass to not support EAPI=7... so pikernel-build and the ebuild might have to update too.
```
* Call stack:
*                           ebuild.sh, line 628:  Called source '/var/db/repos/genpi64/sys-kernel/raspberrypi-kernel/raspberrypi-kernel-5.10.11.ebuild'
*   raspberrypi-kernel-5.10.11.ebuild, line   6:  Called inherit 'pikernel-build'
*                           ebuild.sh, line 308:  Called __qa_source '/var/db/repos/genpi64/eclass/pikernel-build.eclass'
*                           ebuild.sh, line 123:  Called source '/var/db/repos/genpi64/eclass/pikernel-build.eclass'
*               pikernel-build.eclass, line  36:  Called inherit 'kernel-build'
*                           ebuild.sh, line 308:  Called __qa_source '/var/db/repos/gentoo/eclass/kernel-build.eclass'
*                           ebuild.sh, line 123:  Called source '/var/db/repos/gentoo/eclass/kernel-build.eclass'
*                 kernel-build.eclass, line  25:  Called die
* The specific snippet of code:
*      *) die "${ECLASS}: EAPI ${EAPI:-0} not supported" ;;
*  
* If you need support, post the output of `emerge --info '=sys-kernel/raspberrypi-kernel-5.10.11::genpi64'`,
* the complete build log and the output of `emerge -pqv '=sys-kernel/raspberrypi-kernel-5.10.11::genpi64'`.
* Working directory: '/usr/lib/python3.10/site-packages'
* S: '/var/tmp/portage/sys-kernel/raspberrypi-kernel-5.10.11/work/raspberrypi-kernel-5.10.11'
```

jonesmz, 2022-12-31:
I'd be happy to merge a pull request that bumps the EAPI to the newest, if you're interested in working on that.