GenPi64/genpi64-overlay #191: dev-python/rpi-gpio

ShorTie8, 2023-06-19:
PYTHON_TARGETS="-python3_10"isn't working no more.
Thankz
Jeff

jonesmz, 2023-06-20:
I'd be happy to merge a pull request that address this, you'll have to add a new ebuild with the revision number incremented.

Make sure to re-generate the manifest data using the pkgdev program.

samip5, 2025-03-12:
```
RROR: dev-python/rpi-gpio-0.7.0::genpi64 failed (depend phase):
 *   eutils.eclass could not be found by inherit()
 * 
 * Call stack:
 *               ebuild.sh, line 625:  Called source '/var/db/repos/genpi64/dev-python/rpi-gpio/rpi-gpio-0.7.0.ebuild'
 *   rpi-gpio-0.7.0.ebuild, line   8:  Called inherit 'eutils' 'distutils-r1'
 *               ebuild.sh, line 275:  Called die
 * The specific snippet of code:
 *              [[ -z ${location} ]] && die "${1}.eclass could not be found by inherit()"
 * 
 * If you need support, post the output of `emerge --info '=dev-python/rpi-gpio-0.7.0::genpi64'`,
 * the complete build log and the output of `emerge -pqv '=dev-python/rpi-gpio-0.7.0::genpi64'`.
 * Working directory: '/usr/lib/python3.12/site-packages'
 * S: '/rpi-gpio-0.7.0'
```