GenPi64/genpi64-overlay #202: GenPi Image on a Raspberry Pi 5

AvroRJX, 2024-01-20:
**Describe the bug**
Below is an outline of the steps required to enable the GenPi64 image to boot on a Raspberry Pi 5.

**To Reproduce**
Steps to reproduce the behavior:
Mount a current GenPi64 image as chroot, also mount the /boot partition. Enter the chroot.
* create dist-kernel-6.1.63.ebuild by copying and renaming the latest version
* Create the manifest for the 6.1.63 kernel:
ebuild raspberrypi-kernel-6.1.63.ebuild manifest
* Create /etc/portage/package.unmask directory
* Create a file in the above directory containing the line below to unmask the virtual file
=virtual/dist-kernel-6.1.63
* Create a file in /etc/portage/package.keywords containing the lines below to unmask the updated kernel 
sys-kernel/raspberrypi-kernel ~arm64
virtual/dist-kernel-6.1.63 ~arm64
* These steps should now allow you to emerge the 6.1.63 kernel 
emerge -avu —-ignore-default-opts sys-kernel/raspberrypi-kernel
A few hours later you should have the 6.1.63 kernel installed 
* The edit the /boot/config.txt file and add the following lines
[pi5]
kernel=kernel8-p4.img
dtoverlay=vc4-kms-v3d-pi5
max_framebuffers=2

**Expected behavior**
Updated image should boot on a RPi 5

jonesmz, 2024-01-20:
> create dist-kernel-6.1.63.ebuild by copying and renaming the latest version
> Create the manifest for the 6.1.63 kernel: ebuild raspberrypi-kernel-6.1.63.ebuild manifest

In this first line is `dist-kernel-6.1.63` supposed to be `raspberrypi-kernel-6.1.63` ?

AvroRJX, 2024-01-21:
That line is correct but I missed out an important prior step which is to enable the rpi5-branch of the genpi64 overlay in order to make the raspberrypi-kernel-6.1.63 ebuild available. I’ve updated the instructions about to add the necessary steps.