GenPi64/genpi64-overlay #204: media-libs/raspberrypi-userland vs dev-embedded/raspberrypi-utils-0_p20240203

stefand, 2024-03-10:
**Describe the bug**

Since a few days I am getting a package conflict when trying to run emerge -uDN world:

[blocks B      ] media-libs/raspberrypi-userland ("media-libs/raspberrypi-userland" is soft blocking dev-embedded/raspberrypi-utils-0_p20240203)

  (media-libs/raspberrypi-userland-0_pre20201022:0/0::gentoo, ebuild scheduled for merge) pulled in by
    media-libs/raspberrypi-userland required by @profile 

  (dev-embedded/raspberrypi-utils-0_p20240203:0/0::gentoo, installed) pulled in by
    dev-embedded/raspberrypi-utils required by (dev-embedded/rpi-eeprom-21.0_p1:0/0::gentoo, installed) USE="pi4 -pi5" PYTHON_TARGETS="python3_11 -python3_10 -python3_12"
    dev-embedded/raspberrypi-utils required by @selected 

media-libs/raspberrypi-userland is pulled in directly by the genpi64:default/linux/arm64/17.0/genpi64 profile. The profile also pulls in dev-embedded/rpi-eeprom, which needs dev-embedded/raspberrypi-utils, which depends on !media-libs/raspberrypi-userland.

**To Reproduce**

Sync genpi and the gentoo repository to current state.

Run emerge -uDNva world

**Image details (please complete the following information):**

genpi64-overlay commit: eadbc9bccae09bb0c60a2468e58fb61dcaf772de
main gentoo repo commit: bc9bfd98746afe73d7df5190574b402b5220025f

**Additional context**

Did I miss some instructions to switch to a different profile?

My workaround is to remove media-libs/raspberrypi-userland from profiles/targets/genpi64/packages. No idea what exactly it provides, but my RPI mostly serves as a nextcloud and samba server.

jonesmz, 2024-03-10:
I've been very busy lately and have not been able to do much with the project.

This seems like an issue with gentoo adding a conflicting package. I'll try to take a look, but if you have time to investigate and open a pull request that would be helpful.

jonesmz, 2024-03-12:
@stefand  does removing

```
media-libs/raspberrypi-userland
```
from `genpi64/packages b/profiles/targets/genpi64/packages`

and adding the file `genpi64/packages b/profiles/targets/genpi64/package.use/rpi-eeprom`
with contents
```
dev-embedded/rpi-eeprom pi4
```

solve your problem?

Please confirm that the removal of `media-libs/raspberrypi-userland` doesn't remove any programs critical for booting the raspberry pi.

stefand, 2024-03-17:
Making the changes you describe solve the problem. In my case I already had the pi4 use flag set on dev-embedded/rpi-eeprom, I think portage complained about it a few weeks ago.

Regarding booting without raspberrypi-userland: My pi boots fine, but that doesn't mean all pis will boot fine. In my case raspberrypi-userland is already uninstalled, so I can't easily check what is/was in it.

However, https://github.com/raspberrypi/userland says "This repo is ancient and deprecated." and to use raspberrypi/utils instead.

jonesmz, 2024-03-17:
See https://github.com/GenPi64/genpi64-overlay/pull/205

PR #205 (2024-03-17): Fix #204
#### Description
A few sentences describing the overall goals of the pull request's commits.

#### Issues Fixed or Closed by this PR

* Fixes #204
commit 5c3d5eea (2024-03-17): Fix #204 (#205)
closed 2024-03-17 by samip5